grblHAL/plugins #9: VFD does not stop spindle when gcode stopped

rvalotta, 2022-05-30:
I have a HY 1.5K Spindle that works perfectly. I can start and stop the spindle from io sender without any problems.   From time to time when i have to abort the program the spindle does not stop.  IO Sender says its off and the rpm is 0 but its still spinning.  I'm not sure what else you would like to help debug this but will answer any questions and provide any data you need.

terjeio, 2022-05-31:
Is this with the latest version?

rvalotta, 2022-06-12:
Sorry for taking a bit to get back to you life has been crazy

[VER:1.1f.20220325:]
[OPT:VNM0SL2,191,1024,3,0]
[NEWOPT:ENUMS,RT+,HOME,ES,TC,SED,ETH]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:iMXRT1062]
[DRIVER VERSION:220325]
[DRIVER OPTIONS:USB.1]
[BOARD:GRBLHAL2000 - PRINTNC]
[AUX IO:4,1,0,0]
[IP:192.168.5.1]
[PLUGIN:MODBUS v0.12]
[PLUGIN:VFD SELECTOR v0.02]
[PLUGIN:HUANYANG v1 v0.07]
[PLUGIN:KEYPAD v1.3 INTERTEST]
[PLUGIN:Macro plugin v0.02]
[PLUGIN:RGB Indicator Lights v1.0]

andrewmarles, 2022-06-15:
I have seen this before as well, all the way up to current version.  It is consistent when using a VFD spindle.  I have looked for the cause of this in the spindle/vfd code a couple of times but have not found the exact cause.  I have tried adding a spindleSetState call (with null state and 0 rpm) in the VFD plugin to the driver reset function, but this didn't solve the problem.

terjeio, 2022-06-15:
I can confirm that there is a bug in the sequencing of commands on a driver reset when a stop command is issued. I have not managed to fix it yet - will switch to a STM32 driver for easier debugging.

terjeio, 2022-06-16:
[Fix comitted](https://github.com/grblHAL/Plugins_spindle/commit/2a9f46f1f3c963605620e433ec7994507dce3429) - there are changes both in the core and the modbus code. Note that I have only tested with my simulator as I do not have a VFD.