grblHAL/plugins #10: Two quick resets are required to clear  Alarm_Spindle

Meliusja, 2022-07-25:
I am writing a plug in  and when I raise Alarm_Spindle it will not clear unless I click reset in I/O Sender twice within a second or so.    A single click will not clear the alarm.   Is there a call I should make when an alarm condition has gone away after raising an alarm.  Or should a reset clear it and wait to see if it comes back?    My alarm handling closely follows the code in Huanyang plug in.

Thanks,
Jeff

Meliusja, 2022-07-29:
I have found using unlock clears Alarm_Spindle immediately.  I Guess I'm not quite sure when I should use reset vs unlock to clear alarms.
Jeff

terjeio, 2023-02-14:
Alarm 1 (hard limit), 2 (soft limit), 10 (estop) and 17 (motor fault) requires a reset - as these are classified as critical.

When alarm 1 and 2 are raised the message `Reset to continue` is reported, I'll add a similar wording to the alarm 10 and 17 messages in the next build.

closed 2023-02-14 by terjeio