grblHAL/plugins #16: Nextion Display support

drom89, 2022-11-25:
Hi @terjeio , would it be possible to add plugin for Nextion displays ? [https://nextion.tech](https://nextion.tech)
I mean something like when you press a button it sends a Gcode command to serial.print

I played with Nextion [Editor](https://nextion.tech/nextion-editor/#_section1) and it should be quite easy to do that.
But i dont know how to implement it to grblHAL, i want to use it on ESP32..
Displays are not that expensive so maybe it will be viable options for all of us.
Thank you.

terjeio, 2022-11-30:
Yes, it is possible to add a plugin - or perhaps it will be possible to use the existing MPG mode? A protocol has to be selected for the communication (plain serial, ModBus, ...) and you have to decide how to package the data for transfer.
MPG mode uses plain serial and standard grbLHAL format - and I have made [this MPG/DRO](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack) based on it.

drom89, 2022-12-27:
So im using ESP32 on board MKS DLC32 V2.1
And i want to use GPIO 0 and GPIO 4 for serial 2 communication, can i use it ? 
And what is the correct syntax to set MPG enable and RX PINs ?

`#if MPG_MODE_ENABLE
  #ifndef MPG_ENABLE_PIN GPIO_NUM_0
  #error "MPG_ENABLE_PIN must be defined when MPG mode is enabled!"
  #endif
  #ifndef UART2_RX_PIN GPIO_NUM_0
  #error "UART2_RX_PIN must be defined when MPG mode is enabled!"
  #endif
#endif`

terjeio, 2022-12-28:
> And i want to use GPIO 0 and GPIO 4 for serial 2 communication, can i use it ?

Yes, I believe so.

> And what is the correct syntax to set MPG enable and RX PINs ?

See the [cnc_boosterpack_map.h](https://github.com/grblHAL/ESP32/blob/master/main/cnc_boosterpack_map.h) file for how it can be done.

drom89, 2023-01-03:
So i added MPG MODE ENABLED to mks_dlc32_2_0_map.h

> #if MPG_MODE_ENABLE
#define UART2_RX_PIN            GPIO_NUM_4
#define MPG_ENABLE_PIN          GPIO_NUM_0
#endif

It compiles and i can flash firmware no problem
I connected RX and TX pins from display to GPIO0 and GPIO4 

> [VER:1.1f.20221115:]
[OPT:VNMSL,35,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,NOPROBE,MPG,REBOOT,TC,SED,RTC]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:ESP32]
[DRIVER VERSION:221205]
[DRIVER OPTIONS:v4.3.4]
[BOARD:MKS DLC32 2.0]

When i try to print some code to serial from display, i cant see anything in console
Do i need to turn on MPG mode some how ?

terjeio, 2023-01-04:
> Do i need to turn on MPG mode/second serial communication some how ?

Yes, the input is blocked unless the MPG mode pin is in the correct state and the controller allows switching mode. More info [here](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack#how-it-works).

The [keypad plugin](https://github.com/grblHAL/Plugin_keypad) allows switching mode via a real-time command character if configured to read input from the serial port - but I am not 100% sure this is currently possible with the ESP32 driver.