grblHAL/plugins #22: Development of a ATC Plugin [Looking for help]

rvalotta, 2023-11-12:
I've started to write a plugin to work with my ATC Setup...  I've added the code needed to plugins_init and can confirm that I am seeing the report options in the console... I've setup a breakpoint on my functions for hal.tool.change and select and they never seem to trigger...   Am I missing something here?   This is a STM32F401 

`#include "driver.h"

#include "grbl/hal.h"
#include "grbl/protocol.h"
#include "grbl/motion_control.h"

#if ATC == 1

static tool_data_t *current_tool = NULL, *next_tool = NULL;
static on_report_options_ptr on_report_options;

char alignment[1];

static void report_options (bool newopt)
{
    on_report_options(newopt);

    if(!newopt) {
        hal.stream.write("[PLUGIN:ATC v0.01]" ASCII_EOL);
    }        
}

static void atc_tool_select (tool_data_t *tool, bool next)
{

    if(next) 
        next_tool = tool;
    else 
        current_tool = tool;
}

static status_code_t atc_tool_change (parser_state_t *gc_state)
{
    if(current_tool == NULL || next_tool == NULL)
        return Status_GCodeToolError;

    if(current_tool == next_tool)
        return Status_OK;

    if(!sys.homed.mask || sys.homed.mask != sys.homing.mask)
        return Status_HomingRequired;

    hal.stream.write_all("Tool Change Taking Place");
}

void atc_init (void) {
    hal.tool.select = atc_tool_select;
    hal.tool.change = atc_tool_change;

    //settings_register(&setting_details);

    on_report_options = grbl.on_report_options;
    grbl.on_report_options = report_options;
}

#endif`

rvalotta, 2023-11-12:
Got that part figured out... The settings for toolchange were set to normal... Changed it and now it is behaving as I expected.   Question is is possible to add a new tool change setting from a plugin or is that core ?

terjeio, 2023-11-12:
You have to set the ATC capability flag in your init function or your [function pointers will be overwritten](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/tool_change.c#L407-L410).

Plugins can [add settings](https://github.com/grblHAL/Templates/blob/master/my_plugin/settings/my_plugin.c), many do that. And there are a number of setting id's that are [reserved for user plugins](https://github.com/grblHAL/core/blob/04a5edcf8353f7294167d12d32e4b073d9be9ce4/settings.h#L278-L288).

rvalotta, 2023-11-12:
Perfect thank you that has me back on track...

Question if this is going to be a public plugin how do I go about getting setting ids?  Looking at about 31 settings as of now

terjeio, 2023-11-18:
Normally getting new settings ids allocated would be by a pull request or by posting the definitions in a comment.
However, since you want quite a high number of ids I'll just reserve a block of 100 for ATC use for now - 900 to 999.

closed 2023-11-24 by rvalotta