grblHAL/plugins #23: ATC and tool change

rvalotta, 2023-11-24:
At the point where I am ready to have the plugin measure each tool as tool changes happen... I figured I could just set $341=2 and have the core do a automatic tool change...   Tracing the code I get to settings.c set_tool_change_mode(). It says if hal.driver_cap.atc is set to return that it is an invalid statement.   

What is the proper / expected way to have the atc to a TLO

terjeio, 2023-11-24:
> What is the proper / expected way to have the atc do a TLO

By implementing similar code as used by the manual tool change functionality in [tool_change.c](https://github.com/grblHAL/core/blob/master/tool_change.c)?
Or should the probing functionality provided by _tool_change.c_ be somehow made generally available (callable from ATC code)?

terjeio, 2023-11-24:
Another option would be to enable tool table support and store premeasured offsets.

rvalotta, 2023-11-24:
I was hoping I would be able to just set the toolchange to Automatic touch off @ G59.3 and just go iwth that...  The atc had a static location for the touch off plate. I didn't see the need to re-invent the code... I will look at tool_change.c and implement what i need... just seems silly to have the same code in place twice.

terjeio, 2023-12-19:
> just seems silly to have the same code in place twice.

It is not straightforward to expose the touch off code in _tool_change.c_ as it is integrated in the manual tool change workflow. It might be possible to refactor the code in order to expose it though. Or perhaps it will be safer and less work to add a new exposable function in the core and leave the manual tool change code as-is? Retesting code takes time...

closed 2024-01-01 by rvalotta