grblHAL/plugins #24: Are Negative Setting Possible?

rvalotta, 2024-01-01:
I'm trying to setup a Format_Decimal to take a nagative value but it does not seem to allow me to do so.

    { 904, Group_UserSettings, "Pocket 1 X Position", "mm", Format_Int16, "###0.000", "-9999.999", "9999.999", Setting_IsExtended, &my_settings.pocket_1_x_pos, NULL, NULL },

No matter what i try it will not take negative values.

Am I missing something?

terjeio, 2024-01-01:
Use `"-###0.000"` for the format string.

rvalotta, 2024-01-07:
I've tried the above and when I $904=-300 it works as expected, When I open it in iosender and try and set the value I get "Unhandled exception occurred 'value' parameter can not be NaN"

Any suggestions on what I might of missed

terjeio, 2024-01-07:
Post the output from `$904` and `$$=904`.

rvalotta, 2024-01-07:
$904=-1000.000
ok
$$=904
$904: Pocket 1 X Position in mm, range: -9999.999 - 9999.999
Value: X Machine Coordinate (mm)
The x axis position referencing the center of the first tool pocket.
ok

rvalotta, 2024-01-07:
Also this works fine when the format does not have the leading -

terjeio, 2024-01-09:
It is a ioSender bug - I'll upload a edge version with a fix later.

terjeio, 2024-01-10:
New [edge version](http://www.io-engineering.com/downloads) with fix uploded.

rvalotta, 2024-01-11:
Thank you, this worked as expected.

closed 2024-01-11 by rvalotta