grblHAL/plugins #25: Request: Support for ESP32 SPI (Slave) WiFi / Bluetooth

gxdeange, 2024-02-13:
Is it possible to add / create a plugin that allows the use of an ESP32 Module (in Slave mode) to add Wi-Fi / Bluetooth functionality via the SPI port?
Regards
George

terjeio, 2024-02-13:
Yes, it can be done.
The networking plugin has support for some [WizNet ethernet modules](https://github.com/grblHAL/Plugin_networking/tree/master/wiznet) already, which relies upon [driver specific SPI interface](https://github.com/grblHAL/RP2040/blob/master/w5x00_ll_driver.c) implementations on top of the lowest level [SPI code](https://github.com/grblHAL/RP2040/blob/master/spi.c). So something similar has to be added for a new protocol. Bluetooth has to be added to such a stack, I assume for virtual serial port(s).
If the SPI protocol for communicating with the ESP32 has to be defined and implemented then that will add quite a bit to the task.

gxdeange, 2024-02-24:
Hi Terje Io,

I found this site that explains that the ESP32 module can be used as a
slave WiFi / BT slave module;

https://www.espressif.com/en/products/socs/esp32#:~:text=ESP32%20can%20perform%20as%20a,SDIO%20or%20I2C%20%2F%20UART%20interfaces
.

and then I found this GitHub page that I think has the specific firmware
for the ESP32 module;

https://github.com/espressif/esp-at

I am seeing if the guys at CNC3D can make sense of this and if it can be
used to create a slave SPI device that I am hoping you could build the
plugin for.

I can then easily build a WiFi / BT hardware module.

Cheers and have a great weekend
George

terjeio, 2024-02-24:
> I am seeing if the guys at CNC3D can make sense of this 

For what purposes, serial over TCP? Or a full TCP/IP stack?
If they have implemented a lwip low-level driver like the WizNet one then with luck the current networking plugin can be used. If not and a full TCP/IP stack with protocol support on top is required then that will be a major undertaking.

FYI the [Bluetooth plugin](https://github.com/grblHAL/Plugins_Bluetooth/) uses AT commands over UART to establish a serial connection, this could likely be adapted for telnet over UART fairly easily.

gxdeange, 2024-02-25:
I am hoping they can incorporate the Espressif firmware onto an ESP32 module so it can be used as a slave device for other MCU's (such as the STM32 platform). I expect it will be full TCP/IP but I am really not sure.
I am hoping that you can reuse as much code / plugins you already have to simplify this as much as possible (similar communications method as the Networking Plugin)

Cheers
George

terjeio, 2024-02-28:
I'll put this on my longer term todo list since I need a break from intensive grblHAL development.
If someone else want to give it a shot before I do then go for it.

Note that a [WizNet module](https://www.wiznet.io/product-item/wizfi360/) might be a better choice - if there is a lwip low-level driver available for it.

terjeio, 2025-02-18:
ESP-AT has been implemented as a [plugin](https://github.com/grblHAL/Plugins_misc/#esp-at), supports Telnet.

closed 2025-02-18 by terjeio